wireless-rc-adapter/wireless-rc-adapter #1: Does not seem to detect as joystick

ahuxtable, 2016-09-03:
Hi,  thanks for going to the effort of doing this but I can't seem to get it to work.

 can program it fine but it just does not detect as a joystick device in windows (10, 64Bit)

I programmed the device directly with the .hex file from the 'installer.zip' and tried it from Arduino 1.6.5 too but both give the same result.

I am using a Pro mini clone and not a genuine sparkfun one if that makes any difference?
 
I have my RX connected and the calibration works as expected but it just does not detect as joystick.

Am I doing something wrong?

ahuxtable, 2016-09-04:
Scratch that - For some reason it just refuses to detect on my laptop.  Worked fine on 2 other machines.  Great work and thanks!

gregnau, 2018-04-05:
Check now, it has been updated since.

closed 2018-04-05 by gregnau