wireless-rc-adapter/wireless-rc-adapter #6: Did not connect as Joy-HID

vatriani, 2018-04-24:
Win10 doesn't shown up the fresh flashed arduino leonardo as an Joy-Dev.

Arduino IDE 1.6.12
Joystick 2.0.5
EEPROM 2.0.0
PinChangeInterrupt 1.2.6

After changing the constructor for Joystick_ in line 33
`Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 2, 0,
  true, true, true, false, false, true,
  false, false, false, false, false);`
to
`Joystick_ Joystick;`

everythings works fine and the leonardo would apear as an generic Joydev.

gregnau, 2018-06-03:
Have you tried it on other win10 devices? 
I mean is this applies to all win10 in general?

ultimate1112, 2018-09-23:
Some devices don't recognise JOYSTICK_TYPE_MULTI_AXIS. Try JOYSTICK_TYPE_JOYSTICK or JOYSTICK_TYPE_GAMEPAD.

gregnau, 2019-03-12:
Issue needs to be undone?
May i ask the reason you have changed that value in commit 98a17b2cb6dc7049bc17dfe445d82afa47342950 @ultimate1112 ?

ultimate1112, 2019-03-12:
Some Windows computers, ie. Windows 7 and 8, like Joystick and Gamepad better, as it simulates an Xbox controller. Thus, more applications can utilize it.