wireless-rc-adapter/wireless-rc-adapter #9: inconsistent connection

iamearlgrey, 2019-03-25:
First, thank you so much for this project. It is exactly what I hav been looking for and can't believe there isn't already a solution like this out there already.

So I can't seem to get the arduino to stop calibrating consistently. I guess that's what's happening when the two lights on the arduino blink on and off simultaneously. I have turned off calibration in the sketch but that doesn't seem to do anything. Sometimes I get the solid light, sometimes it just keeps blinking and I can't figure out what I am doing that gets it to stop blinking and be readable by the computer. It has worked before and I have used in both the velocidrone dam and the fav feeder demo. I think I am connecting everything and starting everything in the same order every time but I can't figure it out.

I am on MacBook pro retina os 10.14.3
the sim I am using is the velocidrone demo
dx6i transmitter
ore 617xl receiver
ebay pro micro clone

gregnau, 2019-03-25:
You're welcome, i'm glad you like it! I know what you mean, felt the same when i got my first RC transmitter in hand and realized there was no way to sim without cables.

Yes this must be calibration mode and you are most likely still using version v2.0, but there is already a v2.1 released with calibration fix. It had multiple issues, well mostly typos and small bugs, which are fixed now. `CAL_DISABLE` was one of them for sure.

Also pay attention to the number of channels in the configuration, because if there is any unused channels, then calibration not automatically accepting.

In case still experiencing issues, then try to upload with `SERIAL_DEBUG` enabled and take a look at the terminal output. During calibration mode it will periodically output the captured channel min and max values.

iamearlgrey, 2019-03-26:
Ok, so I refreshed with v2.1. (there is a miss-key on line 319 of the joystick.ino, should be ":" instead of ";") The calibration is still running even though the #define CAL_DISABLE is uncommented. Anyway, after I rebooted the board calibrated for maybe 20 seconds and then was done and I could use it. I used it for awhile and then unplugged it and came back later and plugged in again and was able to use again after a 30 second calibration. After I was done using I wanted to check again so I plugged my board in again and this time it wouldn't stop calibrating. I stopped timing after 3 minutes.

So before I posted this, I started playing around by looking at there serial monitor. It took me a minute to figure out that what was happening in calibration mode. I was finally able to get a consistent calibration. The problem is still that it calibrates even when uncommented and that it is only consistent in serial monitor...

Ok, again, playing around before posting. Realizing that the calibration will fix while in serial monitor, I realized that it will only calibrate while serial is listening, either serial monitor or the simulator itself. So I need to have the sim running before calibration begins or while calibration is happening, maybe. though I think i've gotten the calibration to end before opening the sim. Still, I am unable to turn off the calibration function, but at least I think I've gotten it to be consistent.

gregnau, 2019-03-26:
Thanks for noticing the miss-key, it is corrected now. The `CAL_DISABLED` switch fix i have forgot to commit last time, but this is also integrated in the re-release of the version 2.1 i made earlier today. Compilation should go without errors now.

About the calibration issue i have not found any bugs. Tried with a few possible configurations, including yours with 8ch PPM modulation, but it was accepting it fine. Are you sure the configured channels are matching the connected/working number of channels? During the calibration all channels must have valid signal. So in case there is 8 ch's configured and only 6 is active (2 is deaf), then calibration will not auto-accept, means it stays in cal-mode infinitely.

Now that the `CAL_DISABLE` switch is truly working, you should try to set it and the `SERIAL_DEBUG` again. Then take a look at the raw input values of the channels in the serial terminal. They should be between something 800-1000 min and 1900-2200 max.

The serial-debug is designed to wait for terminal to open with purpose. Otherwise it is begins transmitting the data too soon, making it impossible to read in case of `32u4` boards (like the pro micro).
When `SERIAL-DEBUG` is not enabled, then it is not waiting with the calibration, as soon as the board is plugged in it begins (if necessary).

Please post me here in `code quote` a short output of the terminal with calibration enabled and one with disabled. That would clarify the source of the issues. Don't forget to include also a few lines from stick movement values!

iamearlgrey, 2019-03-27:
Whatever you did in the last build fixed all of my issues. Works perfectly so far. Thanks so much for helping me troubleshoot. Here is a capture of serial monitor anyway

with calibration on:

```
Wireless RC Adapter (DEBUG MODE)

CALIBRATION ACTIVE
MAX:	1944	1519	1960	1756	1087	1081	0	0
MIN:	1261	1080	1084	997	1086	1080	2500	2500
	OK 	OK 	OK 	OK 	 X 	 X 	OK 	OK 

ALL CHANNEL MUST BE "OK" TO AUTOSAVE!

MAX:	1261	1080	1084	997	1086	1080	2500	2500	1944	1956	1960	1860
A	0
MIN:	1080	1080	1084	996	1086	1080	2500	2500
	OK 	OK 	OK 	OK 	 X 	 X 	OK 	OK 

ALL CHANNEL MUST BE "OK" TO AUTOSAVE!

MAX:	1080	1080	1084	996	1086	1080	2500	2500	1944	1956	1960	1860	1963	1519	0	0
MIN:	1080	1080	1084	996	1086	1080	2500	2500
	OK 	OK 	OK 	OK 	OK 	OK 	OK 	OK 

CALIBRATION FINISHED
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1962 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1962 | CH6: 1081 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1962 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1506 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1081 | CH7: 0 | CH8: 0 | 
CH1: 1506 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1962 | CH6: 1081 | CH7: 0 | CH8: 0 | 
CH1: 1505 | CH2: 1518 | CH3: 1084 | CH4: 1434 | CH5: 1963 | CH6: 1080 | CH7: 0 | CH8: 0 | 
```
with calibration off, now just jumps right in to stick values:

```
Wireless RC Adapter (DEBUG MODE)

CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1084 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1593 | CH5: 1090 | CH6: 1084 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1451 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1522 | CH3: 1087 | CH4: 1464 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1451 | CH5: 1090 | CH6: 1084 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1523 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1510 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1510 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1084 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1510 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
CH1: 1509 | CH2: 1522 | CH3: 1087 | CH4: 1438 | CH5: 1090 | CH6: 1083 | CH7: 0 | CH8: 0 | 
```

gregnau, 2019-03-27:
Great and thx for the logs.
Though it seems like channel 7-8 are not working at all. Can you confirm?
No surprise why it took a minute to understand the calibration debug, it is printing weird values. I will look into this later today.

I have some questions not relevant to the current topic, but for further development:
Was it necessary to calibrate the Pro Micro on the Mac OS at first time you begin to use it as joystick?

iamearlgrey, 2019-03-27:
Yes, as it turns out, my receiver only outputs 6 channels in either pwm or ppm.

As for calibrating the board, no I did not. I just plugged it in and started flashing.

gregnau, 2019-03-27:
Then you should upload it with `CHANNELS 6`  or commented out (since 6 is default), because it could be a reason why calibration not accepting values and proceeding into joystick mode.

I thought so, thx for the feedback.

closed 2019-03-27 by gregnau