wireless-rc-adapter/wireless-rc-adapter #10: Arduino Uno Original

hmrama, 2019-04-17:
I am using the Arduino uno. I can read the chanel but it not detect as joystick. I need to use any firmware? I have try unjoy but not working, it detect as joystick but not working.

mrlunk, 2019-08-12:
Same question & issue here...
WIndows 10 and Arduino UNO original board.
All the files compled and uploaded fine.
Does not get listed as USB HID device.
Please help :)

Greetings,
Peter Lunk

gregnau, 2019-08-21:
Sorry guys, it is not compatible with the UNO yet. I'm working on a serial based solution to make it compatible with almost really anything. Going to release it later this year.
I suggest you to get a 32u4 based board until then.

closed 2019-08-21 by gregnau

mrlunk, 2019-08-25:
Thanks for the tip ! :)
Ordered a nice Arduino pro micro 5V 16MHz - clone for 8$
And all works fine !!!
:)