wireless-rc-adapter/wireless-rc-adapter #13: Mention channel configuration

skorokithakis, 2019-09-05:
Can you mention channel configuration? It doesn't seem to be TAER, since you mention channel 3 as throttle, so what do we need to set the radio to?

gregnau, 2019-09-05:
I'm in hospital until 17th Sept, after that i can help and develop furthet

skorokithakis, 2019-09-05:
Ooh, get well! I can issue a PR if it'll help.

gregnau, 2020-01-11:
Sorry for the late answer!

Channel config can be found here:
https://github.com/wireless-rc-adapter/wireless-rc-adapter/blob/master/joystick.ino

But it is hard to tell a fixed order of channels, since it is configured different in many applications.
Basically if you have a programmable transmitter, then do the channel config there, and if you have a non-programmable then switch the cables or modify the code.

skorokithakis, 2020-01-12:
I see, thank you. My issue was with documenting which channel is set to which axis, which has nothing to do with the transmitter, it is specific to your code. I'm using PPM, so there's only one wire.

WynandGerberAssured, 2020-02-02:
Hi
 I just want 2 channels.
I have tried various configuration without any luck.
Also when i go to game controller to calibrate all axis is jumping in all directions not sure what am o missing.
Any help will be appreciated.