wireless-rc-adapter/wireless-rc-adapter #14: Change to different pins

oldmanbluntz, 2019-12-09:
Not sure if this is the proper channel.

I purchased a 3rd party pro micro, or recognizes as a Leonardo. I attempted to change the pinout in the code under the pwm tab to correspond to proper pins on the pro micro that thinks its a Leonardo, as they both don't have the same pins. I just want to know if there's anything else that has to be done other than change the pinout numbers in the pwm tab to get this to work. I've changed the order under Leonardo to 9,10,6,7,8,5 to correspond to the proper types of pinout between the two boards. Thanks in advance for your help.

oldmanbluntz, 2019-12-11:
Okay, I have the Pro Micro now recognizing as a pro micro, and the proper bootloader installed as far as i know. I have wired this up, with only 2 channels. It seems to wkr, the code compiles and uploads. the computer recognizes it as the sparkfun, devices and printers shows it as a joystick, but, nothing works in the device properties in windows. is this normal behavior? there isn't a lot of troubleshooting information available. I'm trying to figure this out, but with no input or real troubleshooting tips vailable, it's not easy.

oldmanbluntz, 2019-12-13:
Okay. So, what I have is a pro micro that has a Leonardo bootloader on it.

Am I able to just switch the pins in the code to use 9,10,8,11 in that order to use those 4 pins in that order on the Leonardo part of the code, so that I can get this to work on the pro micro with the Leonardo bootloader on it? I have found the 2 posts you made about this elsewhere which gave a bit more insight into this, such as the LEDs meaning when on and off. 

Sorry for posting now 3 times. I'm racking my brain trying to get this to work on this pro micro with a Leonardo bootloader on it.

gregnau, 2020-01-11:
Yes you have to calibrate it before use, it is also in the wiki.
Until calibration is not done, there will be no joystick output.

oldmanbluntz, 2020-01-11:
I just made an 8 channel pwm to ppm converter with a normal atmega328 I can use with a normal rc receiver. It's all good.

closed 2020-01-11 by oldmanbluntz