wireless-rc-adapter/wireless-rc-adapter #17: how to connect it with phoenix v6.0

Guillaume7499, 2020-05-03:
I can connect it with liftoff but phoenix can't detect it. 
Any suggestion ?

jgilbertfpv, 2021-06-27:
Did you ever figure this out?

ciorceri, 2021-12-24:
I can tell you why is happening, but I cannot tell you how to fix that (without additional costs)
I have this kit : https://www.spektrumrc.com/Products/Default.aspx?ProdID=RTM40R5510 and the provided USB interface will work ONLY with Phoenix and doesn't work with any other games.
I assume that Phoenix works only with that proprietary interface which doesn't emulate a joystick.

woody1329, 2024-09-20:
I already have a setup for emulated joystick, which I've just changed into this config, thinking it would work with Phoenix 6, because it says it works with Phoenix on the home page. But, I don't think it ever could